fiberpunk1/Beam-ESP32 #3: Unable to Mount with Printer Offline

jneilliii, 2022-05-07:
**Describe the bug**
When printer is offline/powered off while Node was in an unmounted state unable to mount SD card for file upload.

**To Reproduce**
Steps to reproduce the behavior:
1. Press unmount in Node to give control to printer.
2. Power off printer.
3. Press mount in Node to take back control of SD card.
4. Get error "NO PRINTER"

**Expected behavior**
Automatic mounting of SD card when printer is offline/powered off. 

**Screenshots**
[image]

Notice printer is offline

**Additional context**
Came across this while working on the OctoPrint plugin. I regularly power off my printer on idle using TPLinkSmartplug plugin. It's also probably pretty common with PSU Control and other power control related plugins.

jneilliii, 2022-05-07:
I'm not super familiar with c programming or the underlying API, but would adding `if(current_usb_status)` [here](https://github.com/fiberpunk1/Beam-ESP32/blob/bfe607682e7dfdeb9b5006f4980788e7e7e06c5e/WifiNode/serverprocess.cpp#L691) and [here](https://github.com/fiberpunk1/Beam-ESP32/blob/bfe607682e7dfdeb9b5006f4980788e7e7e06c5e/WifiNode/serverprocess.cpp#L683) resolve this?

fiber-punk, 2022-05-18:
This error message mainly comes from the unmount command. After umount switches the physical switch to the 3D printer, it will also send the M21 command to the printer to initialize the SD card. But every time a Gcode command is sent, it will check if the printer is connected. If it is not connected, it will prompt "NO PRINTER". I will adjust the code of this block so that this unconnected printer does not affect the physical switching.

dezren39, 2022-07-23:
Thank you, seeing that software issues like these are going to be supported going forward is what makes these hardware projects so valuable.

Hologos, 2022-08-25:
Is it fixed yet?

fiber-punk, 2022-09-28:
fixed in the latest release version

closed 2022-09-28 by fiber-punk