fiberpunk1/Beam-ESP32 #8: Feature Request: Upload file to Node via Prusaclicer

volkmarnissen, 2022-09-23:
**Describe the bug**
I just ordered node to use it instead of octoprint. 
One feature is missing:
Prusaslicer can upload gcode directly to octoprint.
Can Node work similarly?
Node is lightweight and easier to use than octoprint. That's why I'd like to replace it.

The prusaslicer uses octoprints Rest API.
So, this should be possible, if Node provides the minimal needed functions:
GET /api/version (new)
POST /api/files/ (/edit)
(May be some others)
[You'll find the octoprint documentation here ](https://docs.octoprint.org/en/master/api/index.html)
This can easily be tested by installing prusaslicer and configuring the host name for the printer.
The upload function (/edit) in Node API has the similar functionality. May be it's enough to add
the /api/version and provide an  additional URI to the /edit function.

 - OS: [e.g. Ubuntu 22.04 64bit]
 - CPU [e.g. Intel i7]
 - Version [e.g. Latest (1.0.9) ]

If you like, I can do the development. It should be only a few lines of code in serverprocess.cpp.
What do you think?

volkmarnissen, 2022-09-30:
Hi,

I saw, that you already implemented the octoprint support. Great!!!
Is it possible to implement also the  additional features of octopis API
in serverprocess::octoFileUpload.
This is:
- gcodeAnalysis and 
- print (start printing after upload)
The gcodeAnalysis makes it very easy to display the progress in the Nodes Web UI.
A very helpful feature for end users!
But this can also  be a later step.

One further remark:
If you make the printer selection configurable in the Web UI rather than having different firmware versions,
You have only one product for all printer versions. Changing printer is then possible without flashing new firmware. 
Then adding new printers is just adding a new class and change the initialization.

Proposal:
Use a PrinterInterface Class and different implementations for Prusa and Marlin.

class PrinterInterface {
  virtual void sendAbortSDprint() =0;
  vrtual void sendBreakAndContinue() =0;
  virtual const char *getPrinterType() const;
...
};
class MarlinPrinter: PrinterInterface {
void  sendAbortSDprint(); {
    sendCmdByPackage("M524\n");
    delay(50);
    sendCmdByPackage("M118 @Stop\n");
M108
};
void  sendBreakAndContinue(); {
          sendCmdByPackage("M108\n");
};
const char *getPrinterType() const
{  
  return "Marlin";
};
...
};
class PrusaPrinter : PrinterInterface{
void  sendAbortSDprint(); {
    sendCmdByPackage("M603\n");
    espGetSDCard();
};
void  sendBreakAndContinue(); {
          sendCmdByPackage("M602\n"");
};
const char *getPrinterType() const
{  
  return "Prusa";
};
...
};

For new Printers:
class OtherPrinter:PrinterInterface {
...
};

initialization:
Printer &printerInterface =getPrinterInterface();
Printer &getPrinterInterface()
{
 switch( getPrinterInterfaceConfiguration())
 {
    case Marlin: 
        printerInterface = new MarlinPrinter();
    break;
    case Prusa: 
        printerInterface = new PrusaPrinter();
    break;
   case default:
            ??? 
}
}
Usage in serverprocess:
printerInterface.sendAbortSDprint(); 
printerInterface.sendBreakAndContinue(); 

I am looking forward to get the device and try it.

Thank you,

Volkmar

volkmarnissen, 2022-09-30:
You could detect the printer automatically: using M16.
https://marlinfw.org/docs/gcode/M016.html
Then you can implement a generic implementation of the initialization.
class PrinterInterface {
virtual bool printerDetected() =0;
}
PrinterInterface &printers[]= {
 MarlinPrinter(),
 PrusaPrinter();
};

PrinterInterface &getPrinter()
{
    for each( p  in  printers )
        if(.p.printerDetected())
            return p;
   return p[0];
};

volkmarnissen, 2022-09-30:
M16 didn't work but 
You can detect printer using  M115:
Prusa returns FIRMWARE_NAME:Prusa...
Marlin returns FIRMWARE_NAME:Marlin...

fiber-punk, 2022-10-05:
Thank you very much for your advice.

It would be very convenient to configure different printers, which we will probably try in the next version.

There are some hardware differences between the current Prusa version of Node and the Ender version of Node, so switching the Node configuration separately may require the user to make some modifications.

M115 is an excellent command. It tells us a lot. But many Marlin has made their own customization of this instruction, and on some machines, Marlin even removed this instruction... Character analysis of different M115 results is a headache for Node

volkmarnissen, 2022-10-05:
OK. Please close this issue, when topic octoprint support is complete for you.
I'll test it when my device arrives.

closed 2022-10-18 by fiber-punk