fiberpunk1/Beam-ESP32 #9: Node confused by "G29 L0" Gcode command, thinks it means that the print is finished

nuevotango42, 2022-10-08:
**Describe the bug**
The `G29 L0` gcode command results in the the node-webpage reporting that a print is done, and then no further commands are submitted so the print makes no progress.

**To Reproduce**
1. Connect to node attached to an ender with JyersUI installed
2. Enter `G29 L0` in Gcode and hit Send
3. See "Print job finish!" in the website.
4. When you hit OK in the dialog the output console then prints: `Ender3V2:Done.`

 I suspect the issue is that whatever marker is used to detect that a print is finished searches for "Done", but in this case bed leveling is also "done", so the node is confused and stops running the job.

**Output from console**
```
cmd:G29 L0
Ender3V2:Mesh loaded from slot 0
Ender3V2:Done.
Ender3V2:ok
```
**System information:**
Ender 3 v2 with JyersUI:
```
cmd:M115
Node version:2022-9-29-2007
Ender3V2:FIRMWARE_NAME:Marlin bugfix-JyersUI v2.0.1 (Nov  5 2021 21:19:32) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ender-3 V2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Ender3V2:Cap:SERIAL_XON_XOFF:0
Ender3V2:Cap:BINARY_FILE_TRANSFER:0
Ender3V2:Cap:EEPROM:1
Ender3V2:Cap:VOLUMETRIC:1
Ender3V2:Cap:AUTOREPORT_POS:0
Ender3V2:Cap:AUTOREPORT_TEMP:1
Ender3V2:Cap:PROGRESS:0
Ender3V2:Cap:PRINT_JOB:1
Ender3V2:Cap:AUTOLEVEL:1
Ender3V2:Cap:RUNOUT:1
Ender3V2:Cap:Z_PROBE:1
```

Node website: `node-webpage-2022-9-29-2007.1`

Node version: `version:2022-9-29-2007`

vlare3d, 2022-10-08:
Thanks.
The point is th string "Ender3V2:Done."
"Done" will triggle Node's print finsh event. 
Is it possible to add M118 @M29 in front of M29, and then Node can filter the following dat?

nuevotango42, 2022-10-08:
I suspect you meant to add `M118 @G29 L0` where currently it says `G29 L0`. I can do that and it doesn't result in being confused about the job finishing, but it doesn't load the mesh, so it means my bed isn't leveled.

jk7gr, 2023-01-07:
> I suspect you meant to add `M118 @G29 L0` where currently it says `G29 L0`. I can do that and it doesn't result in being confused about the job finishing, but it doesn't load the mesh, so it means my bed isn't leveled. Serial output is:
> 
> ```
> cmd:M118 @G29 L0
> Ender3V2:@G29 L0
> Ender3V2:ok
> ```
> 
> I'm pretty sure this is doing nothing, since when I tried adding M118 @ to the `G29 J2` command (probe the bed to determine tilt) it doesn't do anything.

Did you manage to find a resolution on this? I am facing the same issue.
Thank you!

nuevotango42, 2023-01-09:
No, I stopped using node because of this and more frustratingly the terrible flakey serial connection and sd card interface. I found it to be unusable, and when I looked into it found that it seemed to be hardware issues with both so I got rid of it.

fiber-punk, 2023-02-04:
Have fix iin the v1.0.9, this bug is caused by G29 also returning "Done". In fact, the correct way is to let Node Pro use "Done Printing file" as the sign of the end of printing

jk7gr, 2023-02-06:
> Have fix iin the v1.0.9, this bug is caused by G29 also returning "Done". In fact, the correct way is to let Node Pro use "Done Printing file" as the sign of the end of printing

When are you planning to release the 1.0.9?
thank you!

fiber-punk, 2023-02-07:
hi，please refer here:https://github.com/fiberpunk1/Beam-ESP32/releases/tag/v1.0.9

--- original message ---

When are you planning to release the 1.0.9?

thank you!

—

--- end of original message ---

fiber-punk, 2023-02-09:
we released :https://github.com/fiberpunk1/Beam-ESP32/releases

closed 2023-03-09 by fiber-punk

jk7gr, 2023-10-08:
The issue is still persist on the 1.0.9 on the Node PRO. Am I missing something?